pal-robotics/realsense_gazebo_plugin #3: Unable to diplay pointcloud data in rviz

PranavDhulipala, 2020-01-29:
I tried changing the pointcloud tag to true expecting some data to be published on depth/points but i don't see the topic at all. Digging into the code I couldn't find where pointcloud was being published.
Did I miss something? Any help would be appreciated :)

[image]

saikishor, 2020-01-30:
@PranavDhulipala Thanks for reporting the issue, we are working to fix it. We will update you once it is done.

saikishor, 2020-01-31:
@PranavDhulipala We have fixed the plugin, Please try to pull the changes and check it out. You should be able to visualize the pointcloud.

MoTahoun, 2020-02-12:
@saikishor I am facing the same problem also for kinetic-devel branch. 
[image]
 
There is no topic for depth/points

Thanks in advance

saikishor, 2020-02-12:
@MoTahoun Did you enable the `pointCloud` field in the gazebo plugin info as shown by @PranavDhulipala ?

MoTahoun, 2020-02-14:
@saikishor Yes, I missed to enable it. It's fine now. Thanks

anubhav1772, 2020-02-16:
@PranavDhulipala, could you please share your file where you are adding these settings? Actually I am having some problems while running simulated realsense camera in gazebo.

saikishor, 2020-02-16:
@anubhav1772 You can find the information on the URDF and the respective gazebo xacro files here : https://github.com/pal-robotics-forks/realsense/tree/upstream

SubramanianKrish, 2020-02-21:
@saikishor We're trying to mount the realsense on a husky in the URDF. We added the following gazebo tag set into our husky urdf after including the d435 urdf found in realsense2_description package.

`<xacro:include filename="$(find realsense2_description)/urdf/_d435.urdf.xacro"/>`

```
<sensor_d435 parent="top_plate_link" name="d435_frontCamera" >
    <origin xyz="0.35 0 0.005" rpy="0 0 0" />
  </sensor_d435>
```
```
<gazebo reference= "d435_frontCamera">
    <plugin name="d435_frontCamera_controller" filename="librealsense_gazebo_plugin.so">
      <!-- Plugin parameters -->
      <depthUpdateRate>40</depthUpdateRate>   <!-- Find and set correct value !!! -->
      <infraredUpdateRate>40</infraredUpdateRate>   <!-- Find and set correct value !!! -->
      <depthTopicName>/front_camera/depth/image_rect_raw</depthTopicName>
      <depthCameraInfoTopicName> /front_camera/depth/camera_info</depthCameraInfoTopicName>
      <colorTopicName>/front_camera/color/image_rect_raw</colorTopicName>
      <colorCameraInfoTopicName>/front_camera/color/camera_info</colorCameraInfoTopicName>
      <infrared1TopicName> /front_camera/infra1/image_raw </infrared1TopicName>
      <infrared1CameraInfoTopicName> /front_camera/infra1/camera_info </infrared1CameraInfoTopicName>
      <infrared2TopicName> /front_camera/infra2/image_raw </infrared2TopicName>
      <infrared2CameraInfoTopicName> /front_camera/infra2/camera_info </infrared2CameraInfoTopicName>
      <colorOpticalframeName> d435_frontCamera_color_optical_frame </colorOpticalframeName>
      <depthOpticalframeName>  d435_frontCamera_depth_optical_frame </depthOpticalframeName>
      <infrared1OpticalframeName> d435_frontCamera_infra1_optical_frame </infrared1OpticalframeName>
      <infrared2OpticalframeName>  d435_frontCamera_infra2_optical_frame </infrared2OpticalframeName>
      <rangeMinDepth> 0.105 </rangeMinDepth>
      <rangeMaxDepth>  6 </rangeMaxDepth>
      <pointCloud> true </pointCloud>
      <pointCloudTopicName> /front_camera/depth_registered/points</pointCloudTopicName> 
      <pointCloudCutoff> 0.5 </pointCloudCutoff> <!-- copied from defalt values in tutorial -->
      <pointCloudCutoffMax> 3 </pointCloudCutoffMax>
       <!-- <prefix> </prefix> same. what is this? -->
      <!-- <robotNamespace> $(arg robot_namespace) </robotNamespace> -->
    </plugin>
  </gazebo>
```

We're unable to see any of the topics put up being published. Any ideas on where we might be going wrong? I'm using the kinetic-devel branch. Thanks!

P.S: If you'd like me to open this up as a new issue, ping and I will.

PranavDhulipala, 2020-02-24:
@saikishor this fixes the issue of the pointcloud however the point cloud produced doesn't have rgb data.

closed 2020-03-10 by PranavDhulipala

ingramator, 2021-06-08:
@PranavDhulipala Did you have the issue of RGB data solved? If not I will file a new issue because I am not getting rgb data either, just mono for pointcloud.