pal-robotics/realsense_gazebo_plugin #7: No Documentation on How to Use this Package

mzahana, 2020-03-16:
Can you please provide some documentation on how you compile and add the plugin to Gazebo plugin path?

Thank you.

florianspy, 2020-04-04:
Hey i found out.
First download this package and put it into your catkinws src folder and then compile the catkin package.
This will generate a shared library called  librealsense_gazebo_plugin.so for you which is used in the files you gonna download in the next steps.
 you need to download the two xacro files https://github.com/pal-robotics-forks/realsense/tree/upstream/realsense2_description/urdf _d435.gazebo.xacro and  _d435.urdf.xacro
put those two files in the package inside the urdf folder where you want your realsense to be simulated then in the file _d435.urdf.xacro
change the line 13
from
  <xacro:include filename="$(find realsense2_description)/urdf/r435.gazebo.xacro"/>
to packagename with being the name of the package
  <xacro:include filename="$(find packagename)/urdf/r435.gazebo.xacro"/>
Then in the urdf file where you want to use the simulated realsense add the following code, while again replacing packagename with the name of the package you are using. and baselink which the link where you want the realsense camera to be joint on.
  <xacro:include filename="$(find packagename)/urdf/_d435.urdf.xacro" />
  <sensor_d435 parent="${prefix}base_link">
    <origin xyz="0 0 0" rpy="0 0 0"/>
  </sensor_d435>
Afterwards when you launch file which uses this urdf model you will find the simulate realsense and you should see rostopics like /camera/color/image_raw

Kavin-Kailash, 2020-04-25:
My application requires me to add the d435 sensor in a .sdf file containing my main robot description.

Please help me out. I new to modifying description files.

ShrutheeshIR, 2020-04-30:
Hey @florianspy I did the steps you've given here. But there is an error when instantiating the urdf.xacro file. It says 'Not Enough blocks when instantiating macro'. 
I'm fairly new to this, any help would be appreciated thanks!

ShrutheeshIR, 2020-05-10:
This might be a stupid question, but in the _d435.gazebo.xacro file, why is the plugin called librealsense_gazebo_plugin.so ?
Shouldn't it be realsense_gazebo_plugin?
Can you please point out how this filename was created? I went through the CMakeList and there is no mention of librealsense_gazebo_plugin target, only the realsense_gazebo_plugin.
I'm slightly confused. Thanks

florianspy, 2020-05-19:
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">

  <xacro:include filename="$(find ur_e_description)/urdf/ur.transmission.xacro" />
  <xacro:include filename="$(find ur_e_description)/urdf/ur.gazebo.xacro" />

   <xacro:macro name="cylinder_inertial" params="radius length mass *origin">
    <inertial>
      <mass value="${mass}" />
      <xacro:insert_block name="origin" />
      <inertia ixx="${0.0833333 * mass * (3 * radius * radius + length * length)}" ixy="0.0" ixz="0.0"
        iyy="${0.0833333 * mass * (3 * radius * radius + length * length)}" iyz="0.0"
        izz="${0.5 * mass * radius * radius}" />
    </inertial>
  </xacro:macro>

  <xacro:macro name="ur3e_robot" params="prefix joint_limited
    shoulder_pan_lower_limit:=${-pi}    shoulder_pan_upper_limit:=${pi}
    shoulder_lift_lower_limit:=${-pi}    shoulder_lift_upper_limit:=${pi}
    elbow_joint_lower_limit:=${-pi}    elbow_joint_upper_limit:=${pi}
    wrist_1_lower_limit:=${-pi}    wrist_1_upper_limit:=${pi}
    wrist_2_lower_limit:=${-pi}    wrist_2_upper_limit:=${pi}
    wrist_3_lower_limit:=${-pi}    wrist_3_upper_limit:=${pi}
    transmission_hw_interface:=hardware_interface/PositionJointInterface
    safety_limits:=false safety_pos_margin:=0.15
    safety_k_position:=20">

    
    <xacro:property name="base_mass" value="2.0" />  
    <xacro:property name="shoulder_mass" value="2.0" />
    <xacro:property name="upper_arm_mass" value="3.42" />
    <xacro:property name="forearm_mass" value="1.26" />
    <xacro:property name="wrist_1_mass" value="0.8" />
    <xacro:property name="wrist_2_mass" value="0.8" />
    <xacro:property name="wrist_3_mass" value="0.35" />

    
    <xacro:property name="shoulder_cog" value="0.0 -0.02 0.0" />
    <xacro:property name="upper_arm_cog" value="0.13 0.0 0.1157" />
    <xacro:property name="forearm_cog" value="0.05 0.0 0.0238" />
    <xacro:property name="wrist_1_cog" value="0.0 0.0 0.01" />
    <xacro:property name="wrist_2_cog" value="0.0 0.0 0.01" />
    <xacro:property name="wrist_3_cog" value="0.0 0.0 -0.02" />

    
    
    <xacro:property name="d1" value="0.152" />
    <xacro:property name="a2" value="-0.244" />
    <xacro:property name="a3" value="-0.213" />
    <xacro:property name="d4" value="0.104" />
    <xacro:property name="d5" value="0.085" />
    <xacro:property name="d6" value="0.092" />

    
    <xacro:property name="shoulder_offset" value="0.120" />  
    <xacro:property name="elbow_offset" value="-0.093" /> 

    
    <xacro:property name="shoulder_height" value="${d1}" />
    <xacro:property name="upper_arm_length" value="${-a2}" />
    <xacro:property name="forearm_length" value="${-a3}" />
    <xacro:property name="wrist_1_length" value="${d4}" />
    <xacro:property name="wrist_2_length" value="${d5}" />
    <xacro:property name="wrist_3_length" value="${d6}" />

    <link name="${prefix}base_link" >
      <visual>
        <geometry>
          <mesh filename="package://ur_e_description/meshes/ur3e/visual/base.dae"/>
        </geometry>
[... 265 lines cut ...]
    <xacro:ur_arm_gazebo prefix="${prefix}" />

    
    <link name="${prefix}base"/>
    <joint name="${prefix}base_link-base_fixed_joint" type="fixed">
      
      <origin xyz="0 0 0" rpy="0 0 ${-pi}"/>
      <parent link="${prefix}base_link"/>
      <child link="${prefix}base"/>
    </joint>

    
    <link name="${prefix}tool0"/>
    <joint name="${prefix}wrist_3_link-tool0_fixed_joint" type="fixed">
      <origin xyz="0 ${wrist_3_length} 0" rpy="${pi/-2.0} 0 0"/>
      <parent link="${prefix}wrist_3_link"/>
      <child link="${prefix}tool0"/>
    </joint>

  </xacro:macro>
 <xacro:include filename="$(find ur_e_description)/urdf/d435.urdf.xacro" />
<sensor_d435 parent="upper_arm_link">
    <origin xyz="0.0 0.2 0" rpy="0 0 0"/>
  </sensor_d435>
</robot>

callum-sh, 2023-10-25:
Given all this, how can someone simulate the output of the camera as well? Right now, when trying to start the ros node, I keep getting the following:

[realsense2_camera_node-5] [WARN] [1698195013.653152008] [camera.camera]: No RealSense devices were found!