pal-robotics/realsense_gazebo_plugin #8: catkin_make 0% built target

didrif14, 2020-04-15:
when trying to catkin_make the realsense_gazebo_plugin, I get alot of 0% built target.
I have git cloned the repository to my /src folder, and when i catkin_make I get the following error:

```
####
#### Running command: "make cmake_check_build_system" in "/home/megatrond/catkin_ws/build"
####
####
#### Running command: "make -j16 -l16" in "/home/megatrond/catkin_ws/build"
####
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target gazebo_msgs_generate_messages_lisp
[  0%] Built target trajectory_msgs_generate_messages_cpp
[  0%] Built target trajectory_msgs_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_py
[  0%] Built target gazebo_msgs_generate_messages_py
[  0%] Built target gazebo_msgs_generate_messages_nodejs
[  0%] Built target trajectory_msgs_generate_messages_py
[  0%] Built target gazebo_msgs_generate_messages_eus
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target trajectory_msgs_generate_messages_nodejs
[  0%] Built target trajectory_msgs_generate_messages_eus
[  0%] Built target gazebo_msgs_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_generate_messages_eus
[  0%] Built target tf_generate_messages_cpp
[  0%] Built target tf_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_lisp
[  0%] Built target tf_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target actionlib_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target tf_generate_messages_eus
[  0%] Built target gazebo_ros_gencfg
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target std_srvs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target tf_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_srvs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_nodejs
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_py
[100%] Built target realsense_gazebo_plugin
```

v-lopez, 2020-04-15:
This is not a problem, it's just an indicator of the overall progress. 

If you reach 100% and exits without errors you are fine.

didrif14, 2020-04-15:
Are you sure? This comes up everytime I catkin_make. And the camera is not showing up in Gazebo, I figured it might have something to do with this.