pal-robotics/realsense_gazebo_plugin #11: Incorrect Image (lens shown)

ShrutheeshIR, 2020-04-30:
Hello!
I am able to get the camera in my world, and obtain images. However, the images are incorrect. As you can see in the image, the lens of the camera and the outer mesh seems to be overlayed onto the image.  

[image]

I haven't changed the source code. I followed the instructions you provided in another issue [#7 ](https://github.com/pal-robotics/realsense_gazebo_plugin/issues/7)

Please let me know where I am going wrong :)

v-lopez, 2020-04-30:
I believe this should be solved by the `clip` parameters in the gazebo files.

ShrutheeshIR, 2020-04-30:
Hi @v-lopez 
The issue is not with the robot head as you can see. Also, I will be dumping the images directly onto some folder, I'm using rviz just to verify my setup. I need the values to be perfect since I'll be extracting the camera pose to great precision.

v-lopez, 2020-04-30:
Have you checked the clipping? 
Also if you use an rqt_image view to view the image, it appears the same?

ShrutheeshIR, 2020-04-30:
Hey @v-lopez 
Yes, I checked clipping. I tried varying the values and did not work.
I recorded the image as a rosbag, and the images obtained there are completely different as well

ShrutheeshIR, 2020-04-30:
Hi @v-lopez . Update:
Using rqt image_view, I am getting the right images. (Without the problems)
Could you explain why I am getting 3 different outputs? (I am fairly new to ROS and simulations)
I essentially need to extract the images that I got using the image_view, onto files, along with camera poses. How would I go about doing this?

Thanks!

ShrutheeshIR, 2020-05-09:
I was not exactly sure as to what the issue could be. However, when I recorded the images in a rosbag and played it, the images seemed fine. 
Will be closing the issue. Any explanation for the behaviour would be appreciated though :)

closed 2020-05-09 by ShrutheeshIR