pal-robotics/realsense_gazebo_plugin #17: Error while launching multiple cameras mounted on different robots

ahmedlebo, 2020-06-28:
While i am writing the urdf file of my robot i put a sensor d435 like the picture below but whenever i try to make multiple robots out of my robot model this error is raised and only one camera is publishing its data and i can't figure how to modify it to have the 2 cameras on different robots pubish on different topics.
[image]

[image]
[image]
[image]

v-lopez, 2020-06-29:
You should provide `name` parameter like you do with parent: https://github.com/pal-robotics-forks/realsense/blob/kinetic-devel/realsense2_description/urdf/_d435.urdf.xacro#L15

Otherwise it defaults to camera

ahmedlebo, 2020-06-30:
Thanks v-lopez for your reply but When i run view frames this is what i get so it identifies in the d435.urdf.xacro the paramter name and could create the frames correctly but when it comes to the call of the gazebo plugin it doesn't change the camera_name to name that i understand that is the same paramter name creating the names of frames which comes in the prefix paramater and the topics should be with this shape(prefix/topic)
[image]
[image]

[frames.pdf](https://github.com/pal-robotics/realsense_gazebo_plugin/files/4849914/frames.pdf)

ahmedlebo, 2020-07-08:
Please help me with this i am spending quite a while now with so many trials and none works till now

Ahmedkaramakka, 2020-07-11:
@v-lopez I have the same issue , trying to launch multiple robots each with its distinct real sense plugin with no success unfortunately. 
this the launch file used for multiple robots 

<launch>
  
  <param name="robot_description"
    command="$(find xacro)/xacro --inorder $(find robot_gp)/urdf/robot_gp2.urdf.xacro" />

  
  <group ns="robot1">

    <param name="tf_prefix" value="robot1_tf" />

    <include file="$(find robot_gp)/launch/karam_onerobot.launch" >
      <arg name="init_pose" value="-x 1 -y 1 -z 0" />
     <arg name="robot_name"  value="Robot1" />

    </include>
 
  </group>

  
  <group ns="robot2">

    <param name="tf_prefix" value="robot2_tf" />

    <include file="$(find robot_gp)/launch/karam_onerobot.launch" >

      <arg name="init_pose" value="-x -1 -y 1 -z 0" />
      <arg name="robot_name"  value="Robot2" />

    </include>

  </group>

  <group ns="robot3">

    <param name="tf_prefix" value="robot3_tf" />

    <include file="$(find robot_gp)/launch/karam_onerobot.launch" >

      <arg name="init_pose" value="-x 2 -y 1 -z 0" />
      <arg name="robot_name"  value="Robot3" />

    </include>

  </group>

  
  <group ns="robot4">
    <param name="tf_prefix" value="robot4_tf" />
    <include file="$(find robot_gp)/launch/karam_onerobot.launch" >

      <arg name="init_pose" value="-x -2 -y 1 -z 0" />
      <arg name="robot_name"  value="Robot4" />

    </include>
  </group>

</launch>

and this is the file (karam_onerobot.launch)

<launch> 
<arg name="robot_name"/>
 <arg name="init_pose"/>

  
<param name ="/use_sim_time" value="true"/> -->

 <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model"
     args="$(arg init_pose) -urdf -param /robot_description -model $(arg robot_name)"
     respawn="false" output="screen" />

  <node
    name="fake_joint_calibration"
    pkg="rostopic"
    type="rostopic"
    args="pub /calibrated std_msgs/Bool true" />

<node pkg="depthimage_to_laserscan" type="depthimage_to_laserscan" name="depthimage_to_laserscan" >
        <param name="scan_height" value="10"/> 
        <remap from="image" to="camera/depth/color/image_raw" />
        <remap from="camera_info" to="/camera/depth/color/camera_info" />
    </node>

<node pkg="robot_localization" type="ekf_localization_node" name="ekf_localization">
        <rosparam command="load" file="$(find robot_gp)/config/ekf_localization.yaml"/>
    </node>

<node pkg="gmapping" type="slam_gmapping" name="slam_gmapping" output="screen" >
  <remap from="scan" to="/scan" />
  <param name="odom_frame" value="/odom" />
  <param name="base_frame" value="/chasis" />
  <param name="map_update_interval" value="2.0"/>
    <param name="iterations" value="8"/>
    <param name="minimumScore" value="700"/>
    <param name="linearUpdate" value="0.25"/>
    <param name="angularUpdate" value="0.25"/>
</node>

  <node
    name="joint_state_publisher"
    pkg="joint_state_publisher"
    type="joint_state_publisher" />
  <node
    name="robot_state_publisher"
    pkg="robot_state_publisher"
    type="robot_state_publisher"
    output="screen" />  
 <node
    name="rviz"
    pkg="rviz"
    type="rviz"
    args="-d $(find robot_gp)/urdf.rviz" />

  

</launch>

so i am calling the robot urdf each time , inside the robot urdf file i called the gazebo realsense plugin , so I was supposing that it should be repeated with each robot individually like the plugins and laser sensor for an example . ( but this didn't happen) 

inside my robot urdf i called the gazebo plugin using these commands 
<xacro:include filename="$(find robot_gp)/urdf/_d435.urdf.xacro"/>
<sensor_d435>
  parent="chasis" 
  <origin xyz= "-0.000199999999999959 -0.000130000000001091 0.2072" rpy="0 0 0"/>
  name ="${robot_name}"
  topics_ns = "${robot_name}"</sensor_d435>

as you can see I tried to pass the parameters that is asked for in f435 urdf xacro , this didn't work either 

so i also tried these commands instead in my robot  urdf
<xacro:include filename="$(find robot_gp)/urdf/_d435.urdf.xacro"/>
<sensor_d435>
 
  <origin xyz= "-0.000199999999999959 -0.000130000000001091 0.2072" rpy="0 0 0"/>
 
  </sensor_d435>

I assumed that i have no need to pass the parameters like we do in the laser plugin but also this failed 

so any solutions or ideas suggested ?

chongjeelee, 2021-11-17:
i also met the same problem! anyone can solve it ?thanks.

Wayne-xixi, 2021-12-14:
Hi, I have the same issue when I used group and namespace for multiple robot simulation. @Ahmedkaramakka @v-lopez @ahmedlebo @chongjeelee @lucamarchionni Did you guys find a solution? Thanks in advance.