pal-robotics/realsense_gazebo_plugin #18: IMU data

kc944201, 2020-07-14:
Hi

I use Intel D435i for simulation in Gazebo
I want to add IMU data to simulate

Could you offer some instructions to me? 
or some else plugin could suggest

Thanks you!!!

khaledgabr77, 2020-09-07:
i have same issue, if you know how to add imu let me know 
thanks

shonigmann, 2020-11-30:
I was able to add an IMU link and sensor to the D435 URDF output using the standard gazebo plugin. I haven't gotten around to modifying the xacro yet, but in case it helps, the following addition worked for me:

`
  <link name="imu_link">
    <inertial>
      <mass value="0.0005"/>
      <origin xyz="0 0 0"/>
      <inertia ixx="0.00003" ixy="0.0" ixz="0.0" iyy="0.000004" iyz="0.0" izz="0.00003"/>
    </inertial>
  </link> 
  <joint name="imu_joint" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <parent link="camera_link"/>
    <child link="imu_link"/>
  </joint>
  <gazebo reference="imu_link">
    <sensor name="imu_sensor" type="imu">
      <always_on>true</always_on>
      <update_rate>300</update_rate>
      <visualize>true</visualize>
      <topic>__default_topic__</topic>
      <plugin filename="libgazebo_ros_imu_sensor.so" name="imu_plugin">
      	<robotNamespace>camera</robotNamespace>
        <topicName>imu</topicName>
        <bodyName>imu_link</bodyName>
        <updateRateHZ>300.0</updateRateHZ>
        <gaussianNoise>0.0</gaussianNoise>
        <xyzOffset>0 0 0</xyzOffset>
        <rpyOffset>0 0 0</rpyOffset>
        <frameName>camera_link</frameName>
      </plugin>
      <pose>0 0 0 0 0 0</pose>
    </sensor>
  </gazebo>
`

I haven't done too much testing, but this seems to work well enough. Note that, unlike a real D435i, the IMU data will not be synchronized with the image frames.. I haven't found a good way to achieve this yet.

m-tartari, 2020-12-02:
I've made a fork of this repo where I've added gazebo plugins to publish IMU data on the same topics as the Intel D435i.
You can find it as [m-tartari/realsense_gazebo_plugin](https://github.com/m-tartari/realsense_gazebo_plugin), I've put instructions on how to use it in the README file.

v-lopez, 2020-12-02:
@m-tartari If you want to submit a PR we'll happily review it.

At the moment we don't have the resources to add new features to this repo, but we can accept contributions.

m-tartari, 2020-12-02:
@v-lopez I've opened a PR