pal-robotics/realsense_gazebo_plugin #19: Warning: Conversion of sensor type[depth] not supported

HartmannSa, 2020-10-20:
Hi, 
I am using Gazebo 9.0, ros-melodic on an Ubuntu 18.04.5 LTS.
I cloned this plugin in my catkin_ws/src folder and attached the camera to my robot (via urdf files).
In general my world loads and runs, however I always get this warning:

`[Wrn] [msgs.cc:1852] Conversion of sensor type[depth] not supported`

I am concerned, that the plugin is not working correct. Does anyone know what the solution to this warning is?

vincentmaye, 2021-01-17:
I would also be highly interested in a comment on this.

v-lopez, 2021-01-18:
With gazebo 9.11.0-1~bionic on ubuntu 18.04.5 I'm not seeing it.

Are you able to view the RGB, infrared and depth images? If you are I wouldn't give this much more thought.

If not, you could try updating gazebo.