pal-robotics/realsense_gazebo_plugin #31: Align Depth ROS Topic

Drikuy, 2021-04-16:
Hi everyone!

I'm wondering if it would be possible to use `align_depth` topic from original ROS realsense drivers with this package. I need it to use the aligned image of the color camera with the depth picture in Gazebo. 

Thank in advance for any help!

Drojas251, 2021-04-21:
I too am interested in this question. If the align_depth topic is not available, do you have any suggestions for aligning the RGB to Depth image ourselves? 

Thank you!

Drikuy, 2021-04-25:
Hi again,

I'm been searching how to use `aligned_depth_to_color` topics and I found two solutions. First of all, I've had to say that I haven't found any way of publishing directly `aligned_depth_to_color` topics using [pal-robotics/realsense_gazebo_plugin](https://github.com/pal-robotics/realsense_gazebo_plugin) package (please, if anyone knows that I'm wrong, feel free to edit the answer). Nevertheless, the found solutions are exposed below:

1) The first solution is based on hardcoding the files of the [realsense-ros](https://github.com/IntelRealSense/realsense-ros) packages so the field of view and optical frame position parameters from the depth camera of `_d435.urdf.gazebo.xacro` file match with the ones of the color camera. To apply this solution is recomended to clone the package from the repo and start your own branch or , you can just start a fork from the original package. This solution is not really recomended because replaces the `depth` topics for fake `aligned_depth_to_color` topics, but is the fastest solution I've though.

2) The second solution is baed on developing a new ROS node to subscribe to `/color/camera_info`, `/color/image_raw`, `/depth/camera_info` and `/depth/image_raw`. Thus, you can obtain color and not aligned depth images, as well as, the extrinsics and camera calibration parameters via the ROS published information. Then, you can implement an algorithm such as [the following](https://stackoverflow.com/questions/21849512/how-to-align-rgb-and-depth-image-of-kinect-in-opencv) for aligning both pictures. After that, you can publish into your ROS net the new picture in an `aligned_depth_to_color` topic. I did not test this solution because I have no time to do so. However, it seems to me that it is the most accurate way to obtain an `aligned_depth_to_color` image topic using pal-robotics package.

I hope this answer would help other users of this package in the future too!

nickswalker, 2022-04-14:
If you (like me) are thinking: "the plugin publishes an RGB pointcloud, so surely it aligns depth and color internally..." This plugin is based on the Kinect ROS Gazebo plugin, and inherits its code for publishing the point cloud, which as far as I can tell, does not map color over correctly. It acts as though the color data is in depth optical frame and maps it straight on top, ignoring the fact that these cameras are offset. It does not reproject the depth data. I couldn't find anyone directly reporting this issue, but this request was close: https://github.com/ros-simulation/gazebo_ros_pkgs/issues/415

To get both an `aligned_depth_to_color/image_raw` and a `depth/color/points` that has the correct color information, disable the plugin's pointcloud in your URDF, then use depth_image_proc (adjust namespace as appropriate for your setup):
````
<group ns="camera">
  <node pkg="nodelet" type="nodelet" name="nodelet_manager" args="manager" />

  <node pkg="nodelet" type="nodelet" name="nodelet1"
        args="load depth_image_proc/register nodelet_manager">
    <remap from="rgb/camera_info" to="color/camera_info"/>
    <remap from="depth/camera_info" to="depth/camera_info"/>
    <remap from="depth/image_rect" to="depth/image_raw"/>
    <remap from="depth_registered/camera_info" to="aligned_depth_to_color/camera_info"/>
    <remap from="depth_registered/image_rect" to="aligned_depth_to_color/image_raw"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="nodelet2"
        args="load depth_image_proc/point_cloud_xyzrgb nodelet_manager">
    <remap from="rgb/image_rect_color" to="color/image_raw"/>
    <remap from="rgb/camera_info" to="color/camera_info"/>
    <remap from="depth_registered/camera_info" to="aligned_depth_to_color/camera_info"/>
    <remap from="depth_registered/image_rect" to="aligned_depth_to_color/image_raw"/>
    <remap from="depth_registered/points" to="depth/color/points"/>
  </node>
</group>
  ````

The plugin's incorrectly colored point cloud (note the soup can texture mapping onto the background):  
[image]

And using the depth_image_proc to produce the correct registered depth and pointcloud:
[image]

Not point-for-point comparable because the can position is different, but you get the idea.

michael-fenelon, 2025-08-19:
@nickswalker  Thanks for the share. Regarding the line : <remap from="rgb/image_rect_color" to="color/image_raw"/>

Do I have to use imag_proc (http://wiki.ros.org/image_proc) to obtain a rectified image  and input it to nodelet ?

dirodgu, 2025-08-20:
Good morning to both,

I've seen that you are discussing this, and I also want to try to help!

If the displacement is always in the same direction, maybe it has something to do with this parameter of the Gazebo camera plugin `hack_baseline`. This parameter is a virtual offset that is always kept in the Gazebo simulated cameras when set to a value different from 0.

`                <plugin name="${topic_namespace}_camera_controller" filename="libgazebo_ros_camera.so">
                    <camera_name>${topic_namespace}/color</camera_name>
                    <image_topic_name>image_raw</image_topic_name>
                    <camera_info_topic_name>camera_info</camera_info_topic_name>
                    <frame_name>${topic_namespace}_color_optical_frame</frame_name>
                    <hack_baseline>0.00</hack_baseline>
                </plugin>
`

Have you checked if this might be part of the issue? Because this is something completely independent of the calibration of the camera.