pal-robotics/realsense_gazebo_plugin #32: Plan for supporting ROS Noetic / ros release?

HappySamuel, 2021-05-03:
Hi

(1) Is the melodic version adaptable to ROS Noetic as well? Or will there be any noetic release? 
(2) Besides, is there ros release version? that can be install via sudo apt install.

Best,
Samuel

saikishor, 2021-05-03:
Hello,

Thanks for showing interest in the plugin. 

1. Currently, we support Melodic version. however, the plugin does compile in Noetic, but we haven't tested the integration. You could give it a try.
2. No, there is no ROS release version for this repo. You would need to clone and compile this package to use the plugin.

Thanks and Regards,
PAL Robotics Team.

HappySamuel, 2021-05-04:
Hi @saikishor 

(1) Yes, i gave it a try on the noetic yesterday. And it can be compiled, as well as running in gazebo like what it does in Kinetic. Thanks for it.
(2) Hope there'll be ROS release for the plugin, as it is really useful in gazebo simulation. You guys are fantastic.

Best,
Samuel

closed 2021-05-04 by HappySamuel

saikishor, 2021-05-04:
Hello @HappySamuel 

We are glad to know that your issue was solved. Enjoy the plugin.

Thank you.