pal-robotics/realsense_gazebo_plugin #34: realsense plugin for the ignition (gazebo)

addy1997, 2021-05-23:
Hello there,

I have made some changes in the code of this plugin to make it compatible with the [ignition gazebo](https://github.com/ignitionrobotics/ign-gazebo). The source code can be found [here](https://github.com/addy1997/ignition-plugins/tree/dev/realsense_camera) with the [CMakeLists.txt file](https://github.com/addy1997/ignition-plugins/blob/dev/realsense_camera/CMakeLists.txt) to compile it. 

How can I contribute to your project?

v-lopez, 2021-05-26:
If you are still intesrested, you could create a merge request.

I'd have to check how are other gazebo/ignition plugin repositories structured.

addy1997, 2021-05-26:
@v-lopez Yes, I am interested. I have opened the pull request.

addy1997, 2021-06-03:
@v-lopez, any updates?

v-lopez, 2021-06-03:
I've been reviewing how ignition plugins are handled. It looks like they are in completely separated repositories, so it makes no sense to host them here.

I apologise for making you spend time on the PR.

closed 2021-06-03 by v-lopez