pal-robotics/realsense_gazebo_plugin #36: Dark tinted image outputs from image_raw

ingramator, 2021-06-08:
All images from this realsense plugin are noticeable darker than rendered from other Gazebo cameras for some reason.

When loading this plugins xacro into the URDF it also appears to affect ALL other cameras, including non-`realsense_gazebo_plugin cameras`.

Without realsense xacro:
[image]

Then with realsense xacro loaded:
[image]

Note this an identical world with everything identical just with or without gazebo plugin loaded in. The camera is also not occluded by anything.

I have gone through the plugin code and I can't really see anything that looks like it would cause any issues. 

Any ideas?

Cheers

v-lopez, 2021-06-08:
Can't test right now but it could be the camera mesh has some part that is traslucent gray, from the top of my mind there are some parameters for min cutoff distance or something like that.

ingramator, 2021-06-08:
What's even weirder is that if in Gazebo you uncheck Scene -> shadows then, recheck Scene -> shadows it completely resolves the issue. I've played around with min cutoff but doesn't seem to be helping, thanks for the suggestion though since I suspected it may be something like that.

It could also be some weird gazebo lighting bug/interaction.