pal-robotics/realsense_gazebo_plugin #37: No RGB data in output pointcloud

ingramator, 2021-06-09:
When running with pointcloud enable I can only seem to get mono output from the pointcloud output. I am checking RGB in Rviz and manually inspected the cloud output to verify that indeed it is only publishing mono data. 

I can see that the plugin is always dropping into this else if:
https://github.com/pal-robotics/realsense_gazebo_plugin/blob/49ab134fa517280bb8393e715173dc631ccc2557/src/gazebo_ros_realsense.cpp#L185-L190

Since we are filling the `this->image_msg_` data with both RGB colour images and MONO IR images here:
https://github.com/pal-robotics/realsense_gazebo_plugin/blob/49ab134fa517280bb8393e715173dc631ccc2557/src/gazebo_ros_realsense.cpp#L90-L92

I'm suspecting that somehow only the MONO images are actually being used exclusively for:

https://github.com/pal-robotics/realsense_gazebo_plugin/blob/49ab134fa517280bb8393e715173dc631ccc2557/src/gazebo_ros_realsense.cpp#L167

It seems at least one other user may have experienced this issue too: #3 

Any ideas or fixes would be much appreciated! Happy to contribute back if I figure something out too.

Cheers

ingramator, 2021-06-10:
Here's a sample image:

[image]

dantenoguera, 2021-11-10:
could you just have separate image_msgs_ objects for the IR cameras and the RGB camera and work? Would that change the code a lot?

corwin0815, 2021-12-14:
Greating everyone,
I'm recently facing the same problem, so I did the steps john-maidbot provided. And the result stays the same, RGB data does not merged into the pointcloud. Consider that I'm bad at coding and lack of experience, I believe I made some mistakes. Could anyone kindly post the modified code? It will helps me a lot, thanks.

WizKhalista, 2022-04-16:
Hey everyone,
I have the same Problem of not being able to get a coloured Pointcloud. I didn´t get the solution mentioned by john-maidbot to work. Did anyone solve the issue? 
Thank you in advance.