pal-robotics/realsense_gazebo_plugin #53: Simulated D435: PointCloud is shifted on the left wrt real world

torydebra, 2023-01-28:
Hi, 
I found out that there is some error that set the pointcloud a bit on the left wrt the real objects:

[image]

Here you can see the red container, which is positioned according to what gazebo is saying, and the green container which is positioned according to the simulated D435. 

This happens on all objects in any position wrt to the camera. Did anyone else experience this issue?

After some trial and error I found out that the error should be around 0.015m, 
At the moment to remove it I am shifting of this quantity the pointcloud after it is generated by the plugin

michael-fenelon, 2025-08-19:
@torydebra  Any update on the solution ?

torydebra, 2025-08-20:
As I said I am using a hack to just shift the cloud (https://github.com/ADVRHumanoids/realsense_gazebo_plugin/commit/e49178965c7b239692c9945f75f055c9fa8c5818)
but I am not sure anymore that this shift should stay fixed, the error seems to vary according to different scenes, so you should tune it.