pal-robotics/realsense_gazebo_plugin #56: How to install pal-robotics / realsense_gazebo_plugin in ros2

adinathkale01, 2024-04-01:
Please explain the steps how to use above repo in ros2 humble.
I cloned the repo in src folder and did colcon build but it showing following error.
How do i genererate "librealsense_gazebo_plugin.so" file.

--- stderr: realsense_gazebo_plugin                                                                                                                                                             
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< realsense_gazebo_plugin [0.45s, exited with code 1]

SeungsooLee-TIM, 2024-04-03:
I think you tried with the branch melodic-devel. Catkin is a build sytem for ROS1.

And FYI, there seems no version for humble yet.

lss0815, 2024-04-03:
I think you tried with the branch melodic-devel. Catkin is a build sytem for ROS1.

And FYI, there seems no version for humble yet.

MarioCavero, 2024-04-23:
Does it mean this is still not usable in ros2 humble? (Simulating realsense d435 in ros2 and Gazebo , for example).

lss0815, 2024-04-23:
Sorry I forgot fixing. I've also tried last week. It succeeded to get data with the version in humble.

lss0815, 2024-04-25:
I think you've solved yourself well, but FYI I was able to use the plugin only after adding models.

Maybe, could you check the git branch once again? The branch needs to be `foxy-devel`

MarioCavero, 2024-04-25:
Thanks! I fixed it. My situation was adding the camera linked to `base_link`, but not physically to the robot. It was not straightforward as in the tutorials, as the `xacro:sensor` in my case needed to be called different:
from my `robot.urdf.xacro`:

`  <xacro:include filename="$(find mypackage_whatever)/urdf/_d435.urdf.xacro"/>`

  (and in the `_d435.urdf.xacro`, the proper filename for `_d435.gazebo.xacro`
  
  The sensor had to be instantiated like this (probably works without origin?):
```
    <xacro:sensor_d435 name="realsense_d435" topics_ns="realsense_d435" parent="base_link">
  <origin xyz="0.3 0.8 0.8" rpy="0 0.685 -1.57"/>
</xacro:sensor_d435>
```
`ros2 topic list` outputs the correct info, and after reviewing the images, and IRs, all looks good. On rqt view image, depth images are not working, probably because the range needs to be changed as suggested to do in rviz too. The camera cannot be seen visually though but I am not bothered by that.

zp2546265641, 2024-06-27:
> I think you tried with the branch melodic-devel. Catkin is a build sytem for ROS1.
> 
> And FYI, there seems no version for humble yet.
> 
> Edited: foxy branch will work on humble

hello, i download and how to use it in gazebo plugin

lss0815, 2024-06-28:
@zp2546265641 Hello, it might be presumptuous of me, but it would be helpful to refer to the [turtlebot3 simulation](https://github.com/ROBOTIS-GIT/turtlebot3_simulations) and chatGPT when you get started.